rgot-org/TheThingsNetwork_esp32 #4: Error in hal.cpp

kkalbaugh, 2020-02-20:
When I view the serial output I see this:

> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Starting
FAILURE 
C:\Arduino\libraries\TTN_esp32\src\lmic\hal\hal.cpp:393

I tried both a HELTEC ESP32 v1 and v2. 

Any ideas?

enwi, 2020-02-20:
You haven't specified the pins. I know that if you use a Heltec board, it should work right out of the box, but also didn't work for me.

So just use
`ttn.begin(SS, 0xff, RST_LoRa, DIO0, DIO1, DIO2);` instead of `ttn.begin()` in the meantime

kkalbaugh, 2020-02-20:
@enwi Thank you very much.  That was it.

enwi, 2020-02-20:
But this issue should still be investigated as this is not the correct behavior

rgot-org, 2020-02-23:
hello kkalbaugh,
have you selected the right board in the board manager . This error occurs when the auto pin assignement can't  works .
you can see **#pragma message("Board not supported -- use an explicit pinmap")** message in console when compliling.

commit 80618b20 (2020-03-01): Fix #4 due to outdated board defines
closed 2020-03-01 by rgot-org