rgot-org/TheThingsNetwork_esp32 #5: Node activation on TTN

mchacher, 2020-02-21:
Hi
first of all thanks a lot for the lib that is very easy to use.
I am facing an issue with an Heltec LoRa 32 board trying to create my first TTN node.
I am using the example provided trying. TTN is effectively receiving a join request but my device is never getting activated.
Since I am a newbie on LoRaWan I am most probably doing a stupid error. the gateway I am reaching is quite far (1.5 km) from my home. 
When creating my device on TTN I have been struggling with the so-called device EUI. I have used a random one, and the one given by the semtech chip on the board using getDevEui() method from the lib.
Attached is an example of the console from TTN.

Any help would be much appreciated!
[image]

{
  "time": "2020-02-21T17:10:42.887574688Z",
  "frequency": 868.5,
  "modulation": "LORA",
  "data_rate": "SF8BW125",
  "coding_rate": "4/5",
  "gateways": [
    {
      "gtw_id": "eui-7276fffffe019843",
      "timestamp": 136430116,
      "time": "2020-02-21T17:10:41.726077Z",
      "channel": 7,
      "rssi": -120,
      "snr": -3
    }
  ]
}

enwi, 2020-02-23:
In the example you need to specify
```
const char *devEui = "<change this>"; // Change to TTN Device EUI
const char *appEui = "<change this>"; // Change to TTN Application EUI
const char *appKey = "<change this>"; // Change to TTN Application Key
```
you get these parameters from the TTN console of your application under devices and the device you want to use.

rgot-org, 2020-03-07:
The level of reception seems to be very low (-120dBm). Have you tried bringing the node closer to the gateway. Have you a good antenna ?

mchacher, 2020-03-08:
Hi all, I confirm the node was too far from a TTN gateway. It is ok now, I have a gateway at home. ;-)

closed 2020-03-08 by mchacher