rgot-org/TheThingsNetwork_esp32 #6: ttn-otaa with Wireless Stick not working

Bravd42, 2020-02-22:
Hello,

thank you for your simple to use library. I'm using it with my Heltec Ẁifi Lora 32 v2. And it works like a charm. But using it with my Heltec Wireless Stick is not possible. My gateway occasionally receives a Packet but the node is not able to join.

I tried the ports as mentioned in your example:

// wireless stick pinout
#define UNUSED_PIN 0xFF
#define SS 18
#define RST_LoRa 14
#define DIO0 26
#define DIO1 33
#define DIO2 32
...
ttn.begin(SS, UNUSED_PIN, RST_LoRa, DIO0,DIO1,DIO2);

and looking at the pinout sheet from Hetec, I also tried 

#define DIO1 35
#define DIO2 34

but no dice.

Why mustend I provide MISO and MOSI as well? Or am I missing something.

Best Regards,
Bravd

rgot-org, 2020-02-23:
Hello Bravd,
thank you for your support. 
with my wireless stick it works like charm to. 
is you wireless stick on the good frequency ?
Please can you send me your code so i can try ?
Best regards
Francois

Bravd42, 2020-02-24:
Hello,

thanks for the fast reply.

I ordered a secound device. Turns out the first one has a faulty RF unit.
So no issue with your code.

Just one question: Why do the  DIO1 and DIO2 pins in your example don't match the pinout diagram?

Thank you so much for your lib. I finally found a LoRa device and proper ttn support  witch fits all my needs.

Regards,
Bravd

closed 2020-02-24 by Bravd42