rgot-org/TheThingsNetwork_esp32 #9: ttn not connected

sebakiller2018, 2020-03-15:
Hi rgot-org,
first you make grate job with your lib.
i have a problem with the heltec wireless stick connection to the ttn:

my code in from ino file:

#include <TTN_esp32.h>

#include "TTN_CayenneLPP.h"
/***************************************************************************
 *  Go to your TTN console register a device then the copy fields
 *  and replace the CHANGE_ME strings below
 ****************************************************************************/
//const char* devEui = "CHANGE_ME"; // Change to TTN Device EUI
//const char* appEui = "CHANGE_ME"; // Change to TTN Application EUI
//const char* appKey = "CHANGE_ME"; // Chaneg to TTN Application Key

const char *devEui = "1212121212121212"; //changer avec de devEui TTN
const char *appEui = "70B3D57ED002BC1A"; // changer avec le appEui TTN
const char *appKey = "CBFF7FE560C4F6D3140DC4E943C91522"; // changer avec le appKey TTN
// wireless stick pinout
#define UNUSED_PIN 0xFF
#define SS 18
#define RST_LoRa 14
#define DIO0 26
#define DIO1 35
#define DIO2 34

TTN_esp32 ttn ;
TTN_CayenneLPP lpp;

void message(const uint8_t* payload, size_t size, int rssi)
{
    Serial.println("-- MESSAGE");
    Serial.print("Received " + String(size) + " bytes RSSI=" + String(rssi) + "db");
    for (int i = 0; i < size; i++)
    {
        Serial.print(" " + String(payload[i]));
        // Serial.write(payload[i]);
    }

    Serial.println();
}

void setup()
{
    Serial.begin(115200);
    Serial.println("Starting");
    //ttn.begin();
    ttn.begin(SS, UNUSED_PIN, RST_LoRa, DIO0,DIO1,DIO2);
    ttn.onMessage(message); // Declare callback function for handling downlink
                            // messages from server
    ttn.join(devEui, appEui, appKey);
    Serial.print("Joining TTN ");
    while (!ttn.isJoined())
    {
        Serial.print(".");
        delay(500);
    }
    Serial.println("\njoined !");
    ttn.showStatus();
}

void loop()
{
    byte data[1];
    data[0] = 1;
    static float nb = 18.2;
    // nb += 0.1;
    // lpp.reset();
    // lpp.addTemperature(1, nb);
    // if (ttn.sendBytes(lpp.getBuffer(), lpp.getSize()))
    Serial.print(data[0]);
    if (ttn.sendBytes(data, sizeof(data)))
    {
        Serial.printf("Data: %f TTN_CayenneLPP: %d %x %02X%02X\n", data);
  
//        Serial.printf("Temp: %f TTN_CayenneLPP: %d %x %02X%02X\n", nb, lpp.getBuffer()[0], lpp.getBuffer()[1],
//            lpp.getBuffer()[2], lpp.getBuffer()[3]);
    }
    delay(10000);
}
----------------------------------------------------

the serial monitor show only :

checkKeys] provisioned, no session 
Using stored keys to join
3708: [Event] JOINING
Joining TTN ...........316640: [Event] TXSTART
............710441: [Event] JOIN_TXCOMPLETE
............................................................................................................................4593451: [Event] TXSTART
.............4987253: [Event] JOIN_TXCOMPLETE
....

I send one byte wir number 1, but no connection to ttn

can you help me?

thanks 

Sebastian

enwi, 2020-03-15:
The event `JOIN_TXCOMPLETE` suggests that you are not receiving the `Join Accept` downlink message. This issue can occur when using a single channel gateway, a TTIG (with not much traffic) or antenna/region config issues. Does any of this apply?

sebakiller2018, 2020-03-16:
Hi Enwi, the antenna is Standard 868 mz. I Take a Dragon lorahut in raspi this is a 1 Channel hat

rgot-org, 2020-03-16:
Hello sebakiller2018.
Lorawan does not work very well with a 1 channel gateway, it is not intended for.

enwi, 2020-03-16:
Is it [this one](https://wiki.dragino.com/index.php?title=Lora/GPS_HAT) with the SX1276? Also take a look at [this](https://www.thethingsnetwork.org/forum/t/dragino-problems-and-solutions-topic-part-1/8951/180). [Also as @rgot-org mentioned single channel gateways are not supported by TTN](https://www.thethingsnetwork.org/forum/t/single-channel-packet-forwarders-are-deprecated-and-not-supported-guidelines/31117)

closed 2020-03-22 by rgot-org