rgot-org/TheThingsNetwork_esp32 #12: Strange content in received message buffer.

michael-bartmann, 2020-05-27:
The code worked almost immediately with my Heltec V2 board without having to adjust pin mapping.
Both OTA and ABP work. Uplink payloads are delivered to the TTN console just fine. But downlink messages are not. The registered onMessage function is called, but it points to some other stuff. When I receive large messages, the real start is sometimes 9 bytes further in the buffer, seemingly hidden after other (header?) fields.

michael-bartmann, 2020-05-28:
I do not really understand the buffer offsets. I am more a Java developer. But it I change line #1002 of TTN_esp32.cpp to:
std::copy(LMIC.frame + 9, LMIC.frame + 9 + LMIC.dataLen, downlink);
the receiving of Downlink messages works just fine.
So maybe LMIC.frame is nit the right field offset? Works for me...

rgot-org, 2020-05-31:
add 9 bytes offset in mac frame to get payload.

closed 2020-05-31 by rgot-org