rgot-org/TheThingsNetwork_esp32 #14: Send confirmedData

bobhydrm, 2020-06-03:
Hi. I am testing TTN_esp32 with TTGO LoRa. I have problem with sending ConfirmedData from node to network.

call "sendBytes( payload, size, fport, true )" still sends unconfirmed message. Seen as UnconfirmedUplink on gateway and on  network server. 

Everything other works perfectly (sending unconfirmed messages, receiving confirmed and unconfirmed messages from network server).

lrodorigo, 2020-07-10:
Same issue here

Marcel2508, 2020-11-27:
I'm trying to send data and get a confirmation if it reached the network.
I'm sending data with this method-call: ``ttn.sendBytes(DATA,4,1,true)``

Is there any possibility to know if my data got confirmed in the arduino program? 
Because when I turn off my Gateway the method doesn't return false for not getting an ack.

Any Idea how I could implement something to check whether the packet reached it's destination?