rgot-org/TheThingsNetwork_esp32 #15: does it work with M5Stack or ttgo t-beam?

geronimoooooooo, 2020-07-01:
Hey,
would this library also work with the ESP32 M5Stack or the ttgo t-beam?

geronimoooooooo, 2020-07-07:
I tested it with t-beam, and it works.

henkiejan1, 2021-01-15:
No, does not work with the T-beam... I try for a while and get continiously:

[code]
/home/user/Arduino/libraries/TTN_esp32/src/lmic/hal/getpinmap_thisboard.cpp: In function 'const TTN_esp32_LMIC::HalPinmap_t* TTN_esp32_LMIC::GetPinmap_ThisBoard()':
/home/user/Arduino/libraries/TTN_esp32/src/lmic/hal/getpinmap_thisboard.cpp:73:64: note: #pragma message: Board not supported -- use an explicit pinmap
 #pragma message("Board not supported -- use an explicit pinmap")
[/code]

In the sketch is the only thing i read to do is to copy and paste the CHANGME codes.

rgot-org, 2021-03-29:
T-Beam as correctly be implemented in esp32 1.0 .6 framework. Then it should work now ! 
please update your framework and download the library (install from zip)  and test !

closed 2021-10-03 by rgot-org