rgot-org/TheThingsNetwork_esp32 #18: LORA_IO0/1/2 was not declared in this scope

martin3000, 2020-11-06:
I'm using the board TTGO LoRa 32 OLED.
In the code I have "ttn.begin(SS, LMIC_UNUSED_PIN, RST_LoRa, DIO0,DIO1,DIO2);"

When compiling, I get the error:
```
/home/jms/Arduino/libraries/TTN_esp32/src/lmic/hal/getpinmap_ttgo_TBEAM.cpp:38:21: error: 'LORA_IO0' was not declared in this scope:   PIN_SX1276_DIO0 = LORA_IO0,
/home/jms/Arduino/libraries/TTN_esp32/src/lmic/hal/getpinmap_ttgo_TBEAM.cpp:39:21: error: 'LORA_IO1' was not declared in this scope:    PIN_SX1276_DIO1 = LORA_IO1,
/home/jms/Arduino/libraries/TTN_esp32/src/lmic/hal/getpinmap_ttgo_TBEAM.cpp:40:21: error: 'LORA_IO2' was not declared in this scope:    PIN_SX1276_DIO2 = LORA_IO2,
```

I tried to use 
```
#define LORA_IO0 26
#define LORA_IO1 33
#define LORA_IO2 32
```
but that did not help.
I guess for this board, the 3 constants are not defined in the espressif headers.

rgot-org, 2020-11-11:
i have updated the library, please download the repository as zip file then reinstall library from zip.

martin3000, 2020-11-18:
Works fine, thank you!

closed 2020-11-18 by martin3000

richardjlyon, 2021-06-30:
I'm on version 0.1.1 and still have this issue with board `ttgo-lora32-v1`