rgot-org/TheThingsNetwork_esp32 #19: AU915 - more info and testing required?

coratronics, 2020-11-28:
Hi,

I have been testing this library configured for AU915 today but I am having some issues.

## Background:
- Tested with an 8 channel gateway and loraserver - both working with other devices running a different firmware
- Using OTAA
- I had to select the right subband for AU915 before the join requests could start using the right frequencies.

```
bool TTN_esp32::begin(const Arduino_LMIC::HalPinmap_t* pPinmap)
{
    const bool success = os_init_ex(pPinmap);
    if (success)
    {
        // Reset the MAC state. Session and pending data transfers will be discarded.
        LMIC_reset();

        //select subband
        LMIC_selectSubBand(0);
        
    }
    return success;
}
```

## What works:
- SX1276 properly detected
- The code runs - no crashes
- The device is able to issue a join request - loraserver replies with a join accept
- the lmic loop runs in the background

## What does not work:
- The device does not get the join accept reply - it just keeps waiting for the reply.

## Terminal output [keys/ids changed to XX]
```
rstz0x1 (PO�ERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10900
load:0x40080400,len:6360
entry 0x400806b4
Starting
RXMODE_RSSI
[readNvsValue] Reading devEui with value XXXXXXX
[readNvsValue] Reading appEui with value XXXXXXX
[readNvsValue] Reading appKey with value XXXXXXXXXX
[checkKeys] provisioned, no session
[I][TTN_esp32.cpp:452] restoreKeys(): Dev and app EUI and app key have been restored from NVS storage
[checkKeys] provisioned, no session
[readNvsValue] Reading devEui with value XXXXXXX
[readNvsValue] Reading appEui with value XXXXXXXXX
[readNvsValue] Reading appKey with value XXXXXXXX
[I][TTN_esp32.cpp:424] saveKeys(): Dev EUI, app EUI and app key saved in NVS storage
Using stored keys to join
6029: [Event] JOINING
6145: engineUpdate, opmode=0x4
6341: [Event] TXSTART
6470: TXMODE, freq=917600000, len=23, SF=12, BW=125, CR=4/5, IH=0
Joining TTN ...........................................
```
## Observations & Hypothesis
- Something is happening before a EV_TXCOMPLETE event
- Perhaps another function needs to be used to set the right RX "subband" ?

coratron, 2020-11-28:
@rgot-org 

I figured what was wrong with my own setup... The silkscreen on my dev. board was bad...I had DIO configured as pin 25 when it was pin 26. Everything is now unlocked and running. I will post my progress on here as I may have more insights about AU915.

Merci!

coratron, 2020-11-28:
Ok, next debugging step - it could be an issue with the underlying LMIC library but I thought I'd post here just in case:

## Issue
- The device is unable to process downlinks 

```
1637618: [Event] TXCOMPLETE
1637775: engineUpdate, opmode=0x900
1946510: engineUpdate, opmode=0x908
1946532: [Event] TXSTART
1946557: TXMODE, freq=918000000, len=17, SF=12, BW=125, CR=4/5, IH=0
Packet queued
Temp: 18.400002 TTN_CayenneLPP: 1 67 00B8
2090507: setupRx1 txrxFlags 0x20 --> 01
start single rx: now-rxtime: 2
2091136: RXMODE_SINGLE, freq=926900000, SF=12, BW=500, CR=4/5, IH=0
2110082: Invalid downlink, window=RX1
2152757: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 2
2153385: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
rxtimeout: entry: 2158007 rxtime: 2153381 entry-rxtime: 4626 now-entry: 64 rxtime-txend: 124376
2158079: processRx2DnData txrxFlags 0x2 --> 00
2158165: processDnData_norx txrxFlags 00 --> 20
2158431: [Event] TXCOMPLETE
2158588: engineUpdate, opmode=0x900
2571822: engineUpdate, opmode=0x908
2571846: [Event] TXSTART
2571873: TXMODE, freq=918400000, len=17, SF=12, BW=125, CR=4/5, IH=0
Packet queued
```

coratron, 2020-12-01:
I found out the problem was related to ADR... 
- After disabling ADR, everything started working. 
- For ADR to work I had to tweak the Maximum and Minimum DR on Chirpstack.

[https://github.com/mcci-catena/arduino-lmic/issues/624](url)

coratron, 2020-12-01:
## Conclusion
For AU915 users - make sure you have the right Sub-band selected.