rgot-org/TheThingsNetwork_esp32 #29: BLE_provisioning devEUI change problem

kadirozdinc, 2023-07-11:
Hello firstly thanks for sharing this great resources and example with us, I have an issue with provisioning using mobile app shared.

I opened mobile app and change deveui,appeui,appkey. Appeui and Appkey is changed successfully but Deveui doesn't change somehow, app says "I've got problem!". 

I inspect callback functions, for appeui and appkey, program can branch into these function but cannot into deveui.

What might be the reason ?

rgot-org, 2023-07-14:
Hello, 
I just tested and everything is ok for me.
if you use a copy/paste, can you check that there is not a space at the end of your deveui?
click on "ok key" on the keyboard to validate the recording.
if you are in "core level debug: info", you see [ _**54799][I][TTN_esp32.cpp:352] saveKeys(): [ttn_prov] Dev EUI, app EUI and app key saved in NVS storage**_ each time you validate

kadirozdinc, 2023-07-14:
I tried almost everything but in vain there is no space or sth. like that. I understand and can see log for appeui and appkey but deveui didn't work anyway somehow :/ to solve the problem I try another bluetooth serial app from googlePlay. ( I use samsung note 7 phone)